javieryu/nerf_bridge #4: ros_train.py: error: unrecognized arguments: ros_nerfacto

cchester25, 2023-05-26:
Hello, I have started the SLAM algorithm and played back the rosbag with images. Then, I entered the command 'python ros_train.py ros_nerfacto --data /home/pf/CODE/NERF/nerf_bridge/nsros_config_sample.json', but I received an error message saying 'ros_train.py: error: unrecognized arguments: ros_nerfacto'. The image publishing frequency is 15Hz and the odometry frequency is 10Hz. Also, I have modified the camera intrinsic parameters and topics accordingly in the json file.May I know what are the possible solutions to this issue? Thank you!
[image]

cchester25, 2023-05-26:
"When running the command 'python ros_train.py --data /home/pf/CODE/NERF/nerf_bridge/nsros_config_sample.json' without adding 'ros_nerfacto', the code can be started, but it cannot subscribe to the image topic I published. The published topic exists and has data, and the topic name in the JSON file is consistent with the published one. What could be the reason for this?"

javieryu, 2023-05-26:
I'll have a look at the first part to make sure that everything is setup correctly to call the appropriate method names. As for the topic not subscribing, does the nerfbridge node time out, or is there some other behavior happening? Can you give a little more insight into the system that you are using? If your images and poses are not time synchronized then you may have to make some modifications to the nerfbridge code. I'm planning to add this as a feature when I convert the repository to ROS2.

cchester25, 2023-05-26:
> I'll have a look at the first part to make sure that everything is setup correctly to call the appropriate method names. As for the topic not subscribing, does the nerfbridge node time out, or is [...]

"I am using Ubuntu 20.04 with a CPU-17-13 processor and a GPU-3090ti with CUDA 11.8. I am also using ROS-noetic. Additionally, our image and pose data are not yet time synchronized. The image topic has a frequency of 15Hz, while the pose topic has a frequency of 10Hz."

cchester25, 2023-05-26:
> I'll have a look at the first part to make sure that everything is setup correctly to call the appropriate method names. As for the topic not subscribing, does the nerfbridge node time out, or is [...]

"I am using Ubuntu 20.04 with CPU-17-13 generation, GPU-3090ti, CUDA11.8 and ROS-Noetic. Additionally, our images and poses have not been synchronized in time. The image topic is 15Hz and the pose topic is 10Hz. Below is the output during runtime, which shows that it is waiting for images."
[image]

cchester25, 2023-05-26:
> I'll have a look at the first part to make sure that everything is setup correctly to call the appropriate method names. As for the topic not subscribing, does the nerfbridge node time out, or is [...]

"I am using Ubuntu 20.04 with CPU-17-13 generation and GPU-3090ti, CUDA11.8, ROS-Noetic. Additionally, our image and pose frequencies are the same, with image topic at 15hz and pose topic also at 15hz. Below is the output during runtime, which shows that it is waiting for images."
[image]

cchester25, 2023-05-27:
"I solved the issue of 'ros_nerfacto' not being recognized by modifying the command line: 'python ros_train.py --method_name ros_nerfacto --data /home/pf/CODE/NERF/nerf_bridge/nsros_config_sample.json'. You should add '--method_name' before 'ros_nerfacto'."

javieryu, 2023-05-28:
Okay, so this sounds like a problem with the time synchronization in the nerfbridge ros node. This node subscribes to the pose and image topics and then converts them to the coordinate frame convention of Nerfstudio and inserts them in a pytorch dataset. The problem is that this node expects that the images and poses come in pairs with the same time stamp.

It sounds like in your case you have images and poses that do not have matching time stamps so you'll need to make changes to `nsros/ros_dataloader.py`.  You could use an ApproximateTimeSynchronizer from `message_filters` see http://wiki.ros.org/message_filters#ApproximateTime_Policy.

cchester25, 2023-06-02:
> Okay, so this sounds like a problem with the time synchronization in the nerfbridge ros node. This node subscribes to the pose and image topics and then converts them to the coordinate frame [...]
My image and posestamped timestamps are now consistent, but nerf-bridge still can't work properly, and it's stuck at the waiting for data stage.
[image]

eiyike123, 2023-08-27:
I hav

> Okay, so this sounds like a problem with the time synchronization in the nerfbridge ros node. This node subscribes to the pose and image topics and then converts them to the coordinate frame [...]
I am having issues with the subcriber node of Orbslam2. You didnt write thee code that creates the topic.

javieryu, 2023-09-10:
I've updated the documentation to solve this issue, and it will be merged with main once I've finished migrating to ROS2.

closed 2023-09-10 by javieryu

eiyike123, 2023-09-10:
I would like to know if you have finished migrating ros2. For ros1 I couldn’t get the orslam topics

javieryu, 2023-09-10:
It's still experimental, but yes, see the ros2 branch.

eiyike123, 2023-09-10:
Will it work so that I can switch to
It. What about the orbslam

javieryu, 2023-09-10:
> Will it work so that I can switch to It. What about the orbslam

Since it's still experimental, I am not sure!

Eric-Ho-Matrix, 2023-09-11:
Well, sorry to comment under a closed question but I have two questions highly related to this topic.

1. I also got the question that NSROS keep waiting for image streaming. Any detailed solution to this question?
2. Indeed, my image topic and pose topic publish as 30 Hz and 15 Hz, respectively. I know this definitely cause problem. Do you have any detailed solution to solve this. I noticed you mentioned message_filter, is there any other way to fix this?

To be clear, I use realsense d455 and Vins-fusion to get image and pose, respectively.

eiyike123, 2023-09-11:
> Well, sorry to comment under a closed question but I have two questions highly related to this topic.
>
> 1. I also got the question that NSROS keep waiting for image streaming. Any detailed solution [...]

Are you using ros1 or ros2

Eric-Ho-Matrix, 2023-09-11:
> Are you using ros1 or ros2

I'm using ros1 with ubuntu noetic.

eiyike123, 2023-09-11:
And vin fusion publishes the pose of the images received

Eric-Ho-Matrix, 2023-09-11:
> @Eric-Ho-Matrix Please open a new issue with a description of your problem, and I'd be happy to help you figure this out.

Sure! just open a new issue. Thanks for the help.