javieryu/nerf_bridge #8: [Help] How to export the pointcloud

DapengFeng, 2023-06-29:
@javieryu Thank you for your work. It meets an issue with exporting the pointcloud. I look forward to your favorable reply.
[image]

javieryu, 2023-06-29:
I have very little experience working with the exporter script. From what I can tell you have at least one obvious problem, and that is that you are using the wrong version of Nerfstudio for NerfBridge. It appears that the line that raises an error is from a more recent version of nerfstudio. 

See: https://github.com/nerfstudio-project/nerfstudio/blame/d78af5b5699cb343cb46b5163d464e455524040f/nerfstudio/utils/eval_utils.py#L91

To change to the right version, use the following steps to install nerfstudio:
```
git clone --recurse-submodules git@github.com:javieryu/nerf_bridge.git
cd nerf_bridge/nerfstudio
pip install -e .
```

That said, it is possible that the export script will not work regardless of if you change your nerfstudio version. I haven't tested this functionality, and my guess is that it would likely require modifying the exporter.py script in nerfstudio.

DapengFeng, 2023-07-07:
Thanks for your reply.

closed 2023-07-07 by DapengFeng

joejep, 2024-06-13:
@DapengFeng Have you been able to export the mesh or pointcloud using the ROS1 or ROS2 version of the nerfbridge