javieryu/nerf_bridge #9: The submodules

eiyike123, 2023-07-03:
git submodule add https://github.com/chaconinc/DbConnector

results

(nerfstudio) rover@rover-Precision-3571:~$ git submodule add https://github.com/chaconinc/DbConnector
fatal: not a git repository (or any of the parent directories): .git

javieryu, 2023-07-06:
Could you elaborate on your problem? This seems like a git error, and unrelated to a NerfBridge issue.

javieryu, 2023-08-07:
Apologies, for the slow reply. I am currently in the process of migrating to ROS2, but don't have everything finished yet. 

I haven't actually used these, but I think your best bet would be one of these three options:
- https://docs.openvins.com/
- https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam
- https://github.com/open-shade/orbslam3

Please let me know if you have success with any of these. I will be looking into ISAAC ROS VSLAM sometime in September.

eiyike123, 2023-08-07:
I have problems witth the dependency. I already have the images streaming. But I want to have the repo working

javieryu, 2023-08-07:
Which dependency? Also, to be clear, this repository does not work with ROS2.

eiyike123, 2023-08-07:
I am having issues getting the node for the orb_slam2. When will you able to get the ros2 up

eiyike123, 2023-08-07:
It won’t run effectively but I want to see if I use docker but I am new to docker and it’s taking too much time

javieryu, 2023-08-07:
Sorry, ROS can definitely be challenging. Unfortunately, I can't help you trouble shoot your SLAM implementation. I would encourage you to reach out to the maintainers of the SLAM package that you plan to use about issues with their packages.

eiyike123, 2023-08-07:
My camera is realsense camera and I have been able to stream images. I can switch to ros1 and see if it will work. Did you use orb-slam2 or orb-slam3. Can you please tell me the actual orb-slam repo you ran.

javieryu, 2023-08-07:
The setup that I use is outlined in the README: https://github.com/javieryu/nerf_bridge#our-testing-setup .

eiyike123, 2023-08-07:
If I switch to Ros 1 and use real sense camera d415 will it work

elenacliu, 2023-08-27:
@eiyike123 Have you worked it out? I guess it will work because d415 may support ros1 (according to the legacy branch: https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy)

javieryu, 2023-08-28:
I've been using the https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam package with the d455, and it's been working great. I haven't tried it with NerfBridge yet, but I hope to try it soon, so if you're looking for a ROS2 option then that package may work well.

eiyike123, 2023-08-28:
> I've been using the https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam package with the d455, and it's been working great. I haven't tried it with NerfBridge yet, but I hope to try it soon. If you're looking for a ROS2 option then that package may work well.
I want to try it on ROS noetic can you please give the proocess on how to do that.

javieryu, 2023-08-28:
Sorry, I don't know if they support Noetic. Maybe OpenVINS would be a better option if you're using ROS1? I think they support realsense out of the box.

eiyike123, 2023-08-28:
I dont have issues with the realsense camera it is has a camera node already. Your repository says i should have ROS Noetic.

Thank you

elenacliu, 2023-08-29:
@eiyike123 so you have run this repo successfully with realsense camera and ROS neotic right?

eiyike123, 2023-08-29:
> 

Yes I have done that. I have orb_slam but it’s not showing me he topics

javieryu, 2023-09-10:
I am going to close this issue since it seems to have gone a bit off track. 

@eiyike123 If you're still having trouble running NerfBridge please open a new issue with a clear description of your problem. 

@elenacliu WRT functionality with RealSense, I have a working implementation of NerfBridge that works with the D455 and ROS2 Humble on the ros2 branch in this repo. I'm planning on merging this branch to main in a few days once I tackle a few other issues.

closed 2023-09-10 by javieryu