javieryu/nerf_bridge #11: Quality after exporting pointcloud or mesh

tonhathuy, 2023-08-28:
Great work! It seems that nerfStudio's pointcloud or mesh export does not retain the best quality when viewed in the web viewer. Can you show me the exported mesh file of the demo video? I need them to display on mesh file readers like meshlab or threejs. 
Thanks.

javieryu, 2023-08-28:
I've never did an export of the NeRF in the demo video, it was a screen capture of the viewer! If you're looking for advice with creating better meshes I would recommend posting in the Nerfstudio Discord channel. The folks there will probably know more about tweaking your setup for mesh generation. 

Also, I would recommend testing on a static dataset extracted from a rosbag, and then once you have your settings locked in you can transfer to Nerfbridge.

javieryu, 2023-09-10:
Closing due to inactivity.

closed 2023-09-10 by javieryu