javieryu/nerf_bridge #12: Do you have any idea on how to update camera view in latest nerfstudio?

elenacliu, 2023-08-29:
I try to install the nerfstudio in your repo, but the requirements for torch and cuda might be old, and I cannot successfully install tiny-cuda-nn for it. So the older version cannot run.
So I install the latest nerfstudio, and it can run as well, except the viewer's camera updating logics, because the code `ViewerState` in nerfstudio has changed so much. I haven't found a proper way to update the cameras till now.

javieryu, 2023-09-10:
@elenacliu Check out commit `0fc928d`. This should allow the camera positions to populate over time if you refresh the page. I've tested this on my setup and v0.3.3 of nerfstudio.

javieryu, 2023-10-30:
Closing this for now!

closed 2023-10-30 by javieryu