javieryu/nerf_bridge #14: (NSROS) Waiting for for image streaming to begin ....

Eric-Ho-Matrix, 2023-09-11:
Thanks for building this interesting project and thanks for opening source.

When I try to deploy this with realsense d455 camera with Vins-Fusion for localization. bridge_nerf shows  **(NSROS) Waiting for for image streaming to begin ....**

[image]

Since Vins-Fusion doesn't come up with stamped_pose, I manually modify the source code to make it publish stamped_pose with same freq as image information published by realsense camera node (30 Hz).  (check it out how to modify pose publish rate for VINS_FUSION here https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/92#issuecomment-562021646)

[image]

My environment come up with:
ubuntu noetic
ros1
exact version of Nerfstudio as README

javieryu, 2023-09-11:
It looks like you need to use an ApproximateTimeSynchronizer instead of the regular TimeSynchronizer that is used in the code. This should be pretty easy to implement, just modify the instantiation of the time synchronizer in the `ros_dataloader.py` from a TimeSynchronizer to a ApproximateTimeSynchronizer. You'll also need to add a third argument for the "slop" between the topics.

I've implemented some logic for switching between these two message filters in the ros2 branch, and it should be pretty easy to make the changes in the main branch code. [See here](https://github.com/javieryu/nerf_bridge/blob/db866ba2756aac0bc60b06fbb11b1864d07f4ac1/nerfbridge/ros_dataloader.py#L128)

Eric-Ho-Matrix, 2023-09-12:
Thanks for the prompt ! 
Now it is start training. But the new issue is the training process seems wired, it basically shows nothing, which I feel like my coordinates system may not correct. Dose Nerf_bridge save all collected pictures and their corresponding poses? I want to see the details for debugging

javieryu, 2023-09-13:
Yes, this sounds like a coordinate frame issue to me! I would recommend converting your rosbag to the nerfstudio format, and training on that static data set using just the normal Nerfstudio training pipeline. Unfortunately, I can't find the script that I use to do this for ROS1. It should be pretty straightforward to write something with [https://ternaris.gitlab.io/rosbags/](https://ternaris.gitlab.io/rosbags/) to do that though. Then whatever coordinate transformations that you use to make that work should be easy to integrate into NerfBridge.

javieryu, 2024-02-27:
Closing due to inactivity. Feel free to open again if problems persist.

closed 2024-02-27 by javieryu