javieryu/nerf_bridge #17: Problems related to ros2 execution

pilarvidalaguilar, 2023-10-06:
Hi.  
I am trying to execute nerf-bridge in ros2 using the ros2 branch that was recently posted in this repo. I already have the json filled with the intrinsic camera parameters as well as the topics. I am publishing the topics:
-  "image_topic": "/camera/color/image_raw", that is a message of type sensor_msgs/Image.msg
-  "pose_topic": "/visual_slam/tracking/odometry", that is a message of type nav_msgs/Odometry.msg

Both topics are published at the same time, every second (for now, only for testing purposes). The info of these topics are obtained from the poster dataset in nerfstudio (the json containing camera intrinsics, paths to the images and transformation matrices).

I use ns-train ros-nerfacto --data /path/to/config.json as you recommend in the readme.

It shows "Killed" before even starting to train:
[image]

By doing some logging, I've found out that the process dies when it does the self.pipeline config setup inside trainer.py (inside nerfstudio itself), in the setup function.
This is what the data parser shows in the config:
[image]

Any hint on what could be going on? Could it be related to the version of nerfstudio I'm using? (I just installed it so it should be the newest available). Or maybe could it be related to how I'm launching the process? Or the data?

Thanks :)

javieryu, 2023-10-06:
Hmmm, maybe this is a path specification problem. Are you passing a relative path? Maybe try making it absolute?

pilarvidalaguilar, 2023-10-09:
Hello and thanks for the quick response.
The path to the json I'm using is absolute, and by setting some prints in the ros_dataparser.py, I'm able to see it can extract the info in the json related to the intrinsic parameters of the camera and the name of the topics.

javieryu, 2023-10-09:
Hmmm, could you give a bit more information about your setup? Otherwise, it'll be difficult for me to replicate this error.

pilarvidalaguilar, 2023-10-10:
I will explain my testing setup before executing it in a real situation.

So for the data, I'm using the poster dataset used in nerfstudio. I'm taking the json info regarding the intrinsic camera calibration, the path to each image and the transformation matrix of each image. This would be the example for one of the images as well as its corresponding transformation matrix, and the intrinsic parameters of the camera.
[image]

I wrote a python ros2 node to go through the whole json file publishing the following info into the topics:
- I publish messages of type sensor_msgs/Image.msg to the topic named "/camera/color/image_raw" 
- I publish messages of type nav_msgs/Odometry.msg to the topic named "/visual_slam/tracking/odometry"

Both messages are published at the same time every second.

Now that I'm able to see the data getting published in those two topics, I fill the information on your .json config file:
[image]

Then, I run the node publishing the image and the pose topics and then run:

$ ns-train ros-nerfacto --data ~/workspaces/nerfbridge_ws/nerf_bridge-ros2/nerfbridge_sample_config.json 

And this is what I get:

[image]

If you need more information let me know, and thanks for the help.

javieryu, 2023-10-10:
Can you check if you are running out of RAM or GPU memory?

pilarvidalaguilar, 2023-10-11:
I'm using an NVIDIA GeForce RTX 3080 and an Intel i5 with 16GB of RAM. Those 16GB were not enough. By changing to a more powerful computer it started training! So it is working for now. Thanks for the help

javieryu, 2023-10-11:
Awesome! Glad you got it working!

closed 2023-10-11 by javieryu