javieryu/nerf_bridge #21: No module named 'nerfstudio.data.scene_box'

lvmingzhe, 2023-10-30:
Hi, thanks for your great work.
I`m trying to deploy NeRFBridge on an edge computing Nvidia Jetson AGX Orin 64GB platform.
firstly, I git clone the repo by
```
git clone --recursive https://github.com/javieryu/nerf_bridge.git
```
and then install the nerfstudio module successfully.
However when I continue to work and got some trouble.
To my personal understanding, I need to 
```
roslaunch orb_slam3_ros euroc_mono_inertial.launch
```
```
rosbag play ~/data/MH_01_easy.bag
```
, and then 
```
python ros_train.py ros_nerfacto --data /home/hello/code/nerf_bridge/ns_orb3_euroc.json
```
but I got this error 
```
(nerf_bridge) hello@hello-desktop:~/code/nerf_bridge$ python ros_train.py ros_nerfacto --data /home/hello/code/nerf_bridge/ns_orb3_euroc.json 
/home/hello/mambaforge/envs/nerf_bridge/lib/python3.8/site-packages/tinycudann/modules.py:53: UserWarning: tinycudann was built for lower compute capability (86) than the system's (87). Performance may be suboptimal.
  warnings.warn(f"tinycudann was built for lower compute capability ({cc}) than the system's ({system_compute_capability}). Performance may be suboptimal.")
Traceback (most recent call last):
  File "ros_train.py", line 29, in <module>
    from nsros.method_configs import AnnotatedBaseConfigUnion
  File "/home/hello/code/nerf_bridge/nsros/method_configs.py", line 17, in <module>
    from nerfstudio.models.nerfacto import NerfactoModelConfig
  File "/home/hello/code/nerf_bridge/nerfstudio/nerfstudio/models/nerfacto.py", line 35, in <module>
    from nerfstudio.fields.density_fields import HashMLPDensityField
  File "/home/hello/code/nerf_bridge/nerfstudio/nerfstudio/fields/density_fields.py", line 26, in <module>
    from nerfstudio.data.scene_box import SceneBox
ModuleNotFoundError: No module named 'nerfstudio.data.scene_box'

```

Could you give me some suggestions? Thanks very much.

lvmingzhe, 2023-10-30:
This command works for me:
```
export PYTHONPATH=$PYTHONPATH:/home/hello/code/nerf_bridge/nerfstudio
```

closed 2023-10-30 by lvmingzhe

javieryu, 2023-10-30:
Hi @lvmingzhe, glad you got this working. How is the performance on the AGX? I have one on my desk but haven't had a chance to try it out yet!

lvmingzhe, 2023-11-01:
> Hi @lvmingzhe, glad you got this working. How is the performance on the AGX? I have one on my desk but haven't had a chance to try it out yet!

Hi @javieryu , thanks for your question. I conducted a test using the tum_rgb_freiburg1_xyz dataset. With NerfStudio, I observed a training speed of approximately 35K rays/second. This seems to be roughly one-third of the speed when compared to running on an RTX 3090.

Augusthyq, 2024-06-11:
> ns_orb3_euroc.json

Sorry, how do you get the ns_orb3_euroc.json file? I could not find it.