javieryu/nerf_bridge #22: Issue loading checkpoints

bharath-k1000, 2023-11-06:
Hello!
[image]

While trying to use the viewer to load trained models, it gives an error that the directory does not exist. Even after specifying the checkpoint file directly, it is not able to load. Perhaps it could be a basic error with the directory path.  But the path seems to be correct.
Would appreciate any insight on this !

bharath-k1000, 2023-11-08:
It was path specification(😅)  + having to disable drawing training images and loading train dataset.
The checkpoints are being loaded now.

javieryu, 2023-11-08:
Nice, glad this is solved! Feel free to open a PR with your forked code that does this check pointing.

closed 2023-11-08 by javieryu

SivadineshPonrajan, 2024-04-09:
I could train the ros_nerfacto and save the files. But when I try to export them to pointcloud or mesh, I get the following error. I believe it is something related to checkpoint but I am not sure as well. I am using ROS1 with orbslam3. 

I double-checked that I am using the correct version of nerfstudio as well.

I also tried @bharath-k1000  's nerf_bridge repo for saving the checkpoints and tried to export, still the same error. Any insights could be really helpful. Did you manage to export them @bharath-k1000  ?

`
Loading latest checkpoint from load_dir
✅ Done loading checkpoint from outputs/ros_nerfacto/2024-04-08_225800/nerfstudio_models/step-000029999.ckpt
☁ Computing Point Cloud ☁ ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
{'image': tensor([], size=(0, 1440, 1920, 3)), 'image_idx': tensor([], dtype=torch.int64)}
☁ Computing Point Cloud ☁ ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
☁ Computing Point Cloud ☁ ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% --:--
Traceback (most recent call last):
  File "/root/miniconda3/envs/nerftest/bin/ns-export", line 8, in <module>
    sys.exit(entrypoint())
  File "/root/miniconda3/envs/nerftest/lib/python3.8/site-packages/nerfstudio/scripts/exporter.py", line 463, in entrypoint
    tyro.cli(Commands).main()
  File "/root/miniconda3/envs/nerftest/lib/python3.8/site-packages/nerfstudio/scripts/exporter.py", line 139, in main
    pcd = generate_point_cloud(
  File "/root/miniconda3/envs/nerftest/lib/python3.8/site-packages/nerfstudio/exporter/exporter_utils.py", line 129, in generate_point_cloud
    ray_bundle, _ = pipeline.datamanager.next_train(0)
  File "/root/test/test19/nerf_bridge/nsros/ros_datamanager.py", line 121, in next_train
    batch = self.train_pixel_sampler.sample(image_batch)
  File "/root/miniconda3/envs/nerftest/lib/python3.8/site-packages/nerfstudio/data/pixel_samplers.py", line 274, in sample
    pixel_batch = self.collate_image_dataset_batch(
  File "/root/miniconda3/envs/nerftest/lib/python3.8/site-packages/nerfstudio/data/pixel_samplers.py", line 172, in collate_image_dataset_batch
    collated_batch = {
  File "/root/miniconda3/envs/nerftest/lib/python3.8/site-packages/nerfstudio/data/pixel_samplers.py", line 173, in <dictcomp>
    key: value[c, y, x] for key, value in batch.items() if key != "image_idx" and value is not None
IndexError: index is out of bounds for dimension with size 0                                                   
(nerftest) root@nipg17:~/test/test19/nerf_bridge#
`

javieryu, 2024-04-09:
Unfortunately, I'm no longer actively working with any of the ROS1 implementation. However, in other work I've done similar checkpointing, and these were the steps that were required.

1. Save the camera poses somewhere. Exporting requires the training poses and camera intrinsic, but not the images.
2. Modify the exporter script to load the pipeline with these saved poses.

Sorry I can't go into more detail but the related code is in a codebase that is going to be submitted for publication later this year so I can't release it quite yet!