javieryu/nerf_bridge #24: THE NERF BRIDGE WONT TRain

eiyike123, 2023-11-23:
[image]
[image]

(nerfstudio) root@eiyike-Precision-5820-Tower-X-Series:/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge# python ros_train.py --method_name ros_nerfacto --data /home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros_config_sample.json --pipeline.datamanager.data_update_freq 1.0

optimizer is no longer specified in the CameraOptimizerConfig, it is now defined with the rest of the param groups 
inside the config file under the name 'camera_opt'

/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros/method_configs.py:42: FutureWarning: above message coming from
  camera_optimizer=CameraOptimizerConfig(

CameraOptimizerConfig has been moved from the DataManager to the Model.

/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros/method_configs.py:36: FutureWarning: above message coming from
  datamanager=ROSDataManagerConfig(

CameraOptimizerConfig has been moved from the DataManager to the Model.

/root/miniconda3/envs/nerfstudio/lib/python3.8/site-packages/tyro/_calling.py:242: FutureWarning: above message coming from
  return unwrapped_f(*positional_args, **kwargs), consumed_keywords  # type: ignore
[04:23:25] Using --data alias for --data.pipeline.datamanager.dataparser.data                            ros_train.py:56
──────────────────────────────────────────────────────── Config ────────────────────────────────────────────────────────
ROSTrainerConfig(
    _target=<class 'nsros.ros_trainer.ROSTrainer'>,
    output_dir=PosixPath('outputs'),
    method_name='ros_nerfacto',
    experiment_name=None,
    project_name='nerfstudio-project',
    timestamp='2023-11-23_042325',
    machine=MachineConfig(seed=42, num_devices=1, num_machines=1, machine_rank=0, dist_url='auto', device_type='cuda'),
    logging=LoggingConfig(
        relative_log_dir=PosixPath('.'),
        steps_per_log=10,
        max_buffer_size=20,
        local_writer=LocalWriterConfig(
            _target=<class 'nerfstudio.utils.writer.LocalWriter'>,
            enable=True,
            stats_to_track=(
                <EventName.ITER_TRAIN_TIME: 'Train Iter (time)'>,
                <EventName.TRAIN_RAYS_PER_SEC: 'Train Rays / Sec'>,
                <EventName.CURR_TEST_PSNR: 'Test PSNR'>,
                <EventName.VIS_RAYS_PER_SEC: 'Vis Rays / Sec'>,
                <EventName.TEST_RAYS_PER_SEC: 'Test Rays / Sec'>,
                <EventName.ETA: 'ETA (time)'>
            ),
            max_log_size=10
        ),
        profiler='basic'
    ),
    viewer=ViewerConfig(
        relative_log_filename='viewer_log_filename.txt',
        websocket_port=None,
        websocket_port_default=7007,
        websocket_host='0.0.0.0',
        num_rays_per_chunk=20000,
        max_num_display_images=512,
        quit_on_train_completion=False,
        image_format='jpeg',
        jpeg_quality=90,
        make_share_url=False,
        camera_frustum_scale=0.1,
        default_composite_depth=True
    ),
    pipeline=VanillaPipelineConfig(
        _target=<class 'nerfstudio.pipelines.base_pipeline.VanillaPipeline'>,
        datamanager=ROSDataManagerConfig(
            _target=<class 'nsros.ros_datamanager.ROSDataManager'>,
            data=None,
            masks_on_gpu=False,
            images_on_gpu=False,
[... 141 lines cut ...]
Traceback (most recent call last):
  File "ros_train.py", line 90, in <module>
    entrypoint()
  File "ros_train.py", line 80, in entrypoint
    main(
  File "ros_train.py", line 67, in main
    trainer.setup()
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros/ros_trainer.py", line 50, in setup
    super().setup(test_mode=test_mode)
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nerfstudio/nerfstudio/engine/trainer.py", line 147, in setup
    self.pipeline = self.config.pipeline.setup(
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nerfstudio/nerfstudio/configs/base_config.py", line 54, in setup
    return self._target(self, **kwargs)
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nerfstudio/nerfstudio/pipelines/base_pipeline.py", line 264, in __init__
    self.datamanager: DataManager = config.datamanager.setup(
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nerfstudio/nerfstudio/configs/base_config.py", line 54, in setup
    return self._target(self, **kwargs)
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nerfstudio/nerfstudio/data/datamanagers/base_datamanager.py", line 428, in __init__
    super().__init__()
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nerfstudio/nerfstudio/data/datamanagers/base_datamanager.py", line 189, in __init__
    self.setup_train()
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros/ros_datamanager.py", line 77, in setup_train
    self.train_ray_generator = RayGenerator(
TypeError: __init__() takes 2 positional arguments but 3 were given
(nerfstudio) root@eiyike-Precision-5820-Tower-X-Series:/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge#

javieryu, 2023-11-28:
Are you on the latest version of the main branch?

eiyike123, 2023-11-28:
Yea

eiyike123, 2023-11-28:
I am on the main branch

eiyike123, 2023-11-28:
7 months ago.

eiyike123, 2023-11-28:
ModuleNotFoundError: No module named ‘nerfstudio.configs’

SivadineshPonrajan, 2024-02-27:
```bash
File "/root/nerf_bridge/nsros/ros_datamanager.py", line 77, in setup_train
    self.train_ray_generator = RayGenerator(
TypeError: __init__() takes 2 positional arguments but 3 were given
```

I am facing the same issue as well. is there any solution yet?