javieryu/nerf_bridge #26: Getting error while trying to start training: "Unrecognized arguments: ros_nerfacto"

RomenKlaassen, 2024-01-24:
Hi!

- Cloned the project
- Initialized "Nerfstudio" submodule
- Installed all of Nerfstudio's dependencies
- Executed "pip install -e ." inside the nerfstudio directory
- Run: "python ros_train.py ros_nerfacto --data nsros_config_sample.json"
- Getting parsing error: 
[image]

It seems that the ros_nerfacto method is not recognized.
Does anybody know the fix for this?

Much appreciated!

javieryu, 2024-01-24:
> Executed "pip install -e ." inside the nerfstudio directory

Is this a typo? If you didn't `pip install -e .` in the nerfbridge directory then the method will not register with nerfstudio.

Also, it appears you're using the ROS1 interface. Just FYI I will be archiving that branch and dropping ROS1 support.

RomenKlaassen, 2024-01-26:
@javieryu Thank you for the quick response. Oh I see now... I completely forgot to check out the other ROS2 branch.

Ended up being confused why things didnt run the way I wanted to with Nerfstudio. Seems I was working with the ROS1 
interface!

Never mind me, I will close the issue.

closed 2024-01-26 by RomenKlaassen