javieryu/nerf_bridge #27: ros2 humble with turtlebot3 sim and rosbag

KamalanathanN, 2024-02-09:
i'm trying to test the package with gazebo sim before moving on to actual bot and im facing the following issue
this is my JSON file for sim
```
{
    "camera_model": "PINHOLE",
    "fx": 1696.802685832259,
    "fy": 1696.802685832259,
    "cx": 960.5,
    "cy": 540.5,
    "W": 1920,
    "H": 1080,
    "image_topic": "/camera/image_raw",
    "pose_topic": "/odom"
}
```
[image]
it loads only the first image and a blank white frame in the middle.

#### Things i tried:
- syncing image and the odom topics (did not work) 
- tuning ```update_freq, topic_sync(to approx) and topic_slop```
- custom rosbag data, it loads two images as shown below,
```
{
    "camera_model": "PINHOLE",
    "fx": 532.9841918945312,
    "fy": 532.9841918945312,
    "cx": 550.3889770507812,
    "cy": 303.86602783203125,
    "W": 1104,
    "H": 621,
    "image_topic": "/zed_node/rgb/image_rect_color",
    "pose_topic": "/zed_node/odom"
}
```
[image]

heres the [bag](http://tinyurl.com/zlabscan) i used.

javieryu, 2024-02-09:
Can you verify that more than just the first topic are being received by nerfbridge? You should be able to check this by echoing the dataloader health topic, and then checking if it's increasing.

Also, can you confirm that your poses are not far outside the viewer volume?

KamalanathanN, 2024-02-15:
Hi, i got it to work when i ran with ```--draw_training_images True``` and refresh the page regularly... it seems the frame update doesn't load in real-time, it loads the poses now and starts training but the poses are outside the viewer volume as shown below, i'll try with newer bag and update if it gets resolved, thank you for your quick response :) 

[image]

javieryu, 2024-02-15:
Yeah the need to refresh the page regularly is due to the old Nerfstudio viewer infrastructure. In Nerfstudio 1.0 they upgraded the viewer, and sometime next week NerfBridge will be upgraded to support Nerfstudio 1.0 and the new viewer. I already have it mostly working on this branch https://github.com/javieryu/nerf_bridge/tree/feature/readme-upgrade.

javieryu, 2024-02-27:
Any resolution on this? I am going to close the issue if it's solved.