javieryu/nerf_bridge #28: error during training

eiyike123, 2024-02-27:
optimizer is no longer specified in the CameraOptimizerConfig, it is now defined with the rest of the param groups 
inside the config file under the name 'camera_opt'

/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros/method_configs.py:42: FutureWarning: above message coming from
  camera_optimizer=CameraOptimizerConfig(

CameraOptimizerConfig has been moved from the DataManager to the Model.

/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros/method_configs.py:36: FutureWarning: above message coming from
  datamanager=ROSDataManagerConfig(

CameraOptimizerConfig has been moved from the DataManager to the Model.

/usr/local/lib/python3.8/dist-packages/tyro/_calling.py:245: FutureWarning: above message coming from
  return unwrapped_f(*positional_args, **kwargs), consumed_keywords  # type: ignore
[01:16:16] Using --data alias for --data.pipeline.datamanager.dataparser.data                            ros_train.py:56
──────────────────────────────────────────────────────── Config ────────────────────────────────────────────────────────
ROSTrainerConfig(
    _target=<class 'nsros.ros_trainer.ROSTrainer'>,
    output_dir=PosixPath('outputs'),
    method_name='ros_nerfacto',
    experiment_name=None,
    project_name='nerfstudio-project',
    timestamp='2024-02-27_011616',
    machine=MachineConfig(seed=42, num_devices=1, num_machines=1, machine_rank=0, dist_url='auto', device_type='cuda'),
    logging=LoggingConfig(
        relative_log_dir=PosixPath('.'),
        steps_per_log=10,
        max_buffer_size=20,
        local_writer=LocalWriterConfig(
            _target=<class 'nerfstudio.utils.writer.LocalWriter'>,
            enable=True,
            stats_to_track=(
                <EventName.ITER_TRAIN_TIME: 'Train Iter (time)'>,
                <EventName.TRAIN_RAYS_PER_SEC: 'Train Rays / Sec'>,
                <EventName.CURR_TEST_PSNR: 'Test PSNR'>,
                <EventName.VIS_RAYS_PER_SEC: 'Vis Rays / Sec'>,
                <EventName.TEST_RAYS_PER_SEC: 'Test Rays / Sec'>,
                <EventName.ETA: 'ETA (time)'>
            ),
            max_log_size=10
        ),
        profiler='basic'
    ),
    viewer=ViewerConfig(
        relative_log_filename='viewer_log_filename.txt',
        websocket_port=None,
        websocket_port_default=7007,
        websocket_host='0.0.0.0',
        num_rays_per_chunk=20000,
        max_num_display_images=512,
        quit_on_train_completion=False,
        image_format='jpeg',
        jpeg_quality=75,
        make_share_url=False,
        camera_frustum_scale=0.1,
        default_composite_depth=True
    ),
    pipeline=VanillaPipelineConfig(
        _target=<class 'nerfstudio.pipelines.base_pipeline.VanillaPipeline'>,
        datamanager=ROSDataManagerConfig(
            _target=<class 'nsros.ros_datamanager.ROSDataManager'>,
            data=None,
            masks_on_gpu=False,
            images_on_gpu=False,
            dataparser=ROSDataParserConfig(
                _target=<class 'nsros.ros_dataparser.ROSDataParser'>,
                data=PosixPath('/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros_config_sample.json'),
                scale_factor=1.0,
                aabb_scale=0.8
[... 142 lines cut ...]
Traceback (most recent call last):
  File "ros_train.py", line 90, in <module>
    entrypoint()
  File "ros_train.py", line 80, in entrypoint
    main(
  File "ros_train.py", line 67, in main
    trainer.setup()
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros/ros_trainer.py", line 50, in setup
    super().setup(test_mode=test_mode)
  File "/usr/local/lib/python3.8/dist-packages/nerfstudio/engine/trainer.py", line 149, in setup
    self.pipeline = self.config.pipeline.setup(
  File "/usr/local/lib/python3.8/dist-packages/nerfstudio/configs/base_config.py", line 54, in setup
    return self._target(self, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/nerfstudio/pipelines/base_pipeline.py", line 254, in __init__
    self.datamanager: DataManager = config.datamanager.setup(
  File "/usr/local/lib/python3.8/dist-packages/nerfstudio/configs/base_config.py", line 54, in setup
    return self._target(self, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/nerfstudio/data/datamanagers/base_datamanager.py", line 422, in __init__
    super().__init__()
  File "/usr/local/lib/python3.8/dist-packages/nerfstudio/data/datamanagers/base_datamanager.py", line 181, in __init__
    self.setup_train()
  File "/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros/ros_datamanager.py", line 77, in setup_train
    self.train_ray_generator = RayGenerator(
TypeError: __init__() takes 2 positional arguments but 3 were given
(nerfstudio2) root@eiyike-Precision-5820-Tower-X-Series:/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge#

javieryu, 2024-02-27:
Looks like you're using the wrong version of nerfstudio. Try 0.3.4 instead of 1.0.

eiyike123, 2024-02-27:
is it that same version for ros foxy and humble

eiyike123, 2024-02-27:
(nerfstudio3) root@eiyike-Precision-5820-Tower-X-Series:/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge# python3 ros_train.py --method_name ros_nerfacto --data /home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge/nsros_config_sample.json --pipeline.datamanager.data_update_freq 1.0
[17:40:30] Using --data alias for --data.pipeline.datamanager.dataparser.data                            ros_train.py:56
──────────────────────────────────────────────────────── Config ────────────────────────────────────────────────────────
ROSTrainerConfig(
    _target=<class 'nsros.ros_trainer.ROSTrainer'>,
    output_dir=PosixPath('outputs'),
    method_name='ros_nerfacto',
    experiment_name=None,
    project_name='nerfstudio-project',
    timestamp='2024-02-27_174030',
    machine=MachineConfig(seed=42, num_devices=1, num_machines=1, machine_rank=0, dist_url='auto', device_type='cuda'),
    logging=LoggingConfig(
        relative_log_dir=PosixPath('.'),
        steps_per_log=10,
        max_buffer_size=20,
        local_writer=LocalWriterConfig(
            _target=<class 'nerfstudio.utils.writer.LocalWriter'>,
            enable=True,
            stats_to_track=(
                <EventName.ITER_TRAIN_TIME: 'Train Iter (time)'>,
                <EventName.TRAIN_RAYS_PER_SEC: 'Train Rays / Sec'>,
                <EventName.CURR_TEST_PSNR: 'Test PSNR'>,
                <EventName.VIS_RAYS_PER_SEC: 'Vis Rays / Sec'>,
                <EventName.TEST_RAYS_PER_SEC: 'Test Rays / Sec'>,
                <EventName.ETA: 'ETA (time)'>
            ),
            max_log_size=10
        ),
        profiler='basic'
    ),
    viewer=ViewerConfig(
        relative_log_filename='viewer_log_filename.txt',
        websocket_port=None,
        websocket_port_default=7007,
        websocket_host='0.0.0.0',
        num_rays_per_chunk=20000,
        max_num_display_images=512,
        quit_on_train_completion=False,
        image_format='jpeg',
        jpeg_quality=90,
        make_share_url=False
    ),
    pipeline=VanillaPipelineConfig(
        _target=<class 'nerfstudio.pipelines.base_pipeline.VanillaPipeline'>,
        datamanager=ROSDataManagerConfig(
            _target=<class 'nsros.ros_datamanager.ROSDataManager'>,
            data=None,
            camera_optimizer=CameraOptimizerConfig(
                _target=<class 'nerfstudio.cameras.camera_optimizers.CameraOptimizer'>,
                mode='SO3xR3',
                position_noise_std=0.0,
                orientation_noise_std=0.0,
                optimizer=AdamOptimizerConfig(
                    _target=<class 'torch.optim.adam.Adam'>,
                    lr=0.0006,
                    eps=1e-08,
                    max_norm=None,
                    weight_decay=0.01
                ),
                scheduler=ExponentialDecaySchedulerConfig(
                    _target=<class 'nerfstudio.engine.schedulers.ExponentialDecayScheduler'>,
                    lr_pre_warmup=1e-08,
                    lr_final=None,
                    warmup_steps=0,
                    max_steps=10000,
                    ramp='cosine'
                ),
                param_group='camera_opt'
            ),
            masks_on_gpu=False,
[... 192 lines cut ...]
    self.compile_and_call_fx_graph(tx, list(reversed(stack_values)), root)
  File "/root/minicondaa3/envs/nerfstudio3/lib/python3.8/contextlib.py", line 75, in inner
    return func(*args, **kwds)
  File "/root/minicondaa3/envs/nerfstudio3/lib/python3.8/site-packages/torch/_dynamo/output_graph.py", line 957, in compile_and_call_fx_graph
    compiled_fn = self.call_user_compiler(gm)
  File "/root/minicondaa3/envs/nerfstudio3/lib/python3.8/site-packages/torch/_dynamo/utils.py", line 189, in time_wrapper
    r = func(*args, **kwargs)
  File "/root/minicondaa3/envs/nerfstudio3/lib/python3.8/site-packages/torch/_dynamo/output_graph.py", line 1024, in call_user_compiler
    raise BackendCompilerFailed(self.compiler_fn, e).with_traceback(
  File "/root/minicondaa3/envs/nerfstudio3/lib/python3.8/site-packages/torch/_dynamo/output_graph.py", line 1009, in call_user_compiler
    compiled_fn = compiler_fn(gm, self.example_inputs())
  File "/root/minicondaa3/envs/nerfstudio3/lib/python3.8/site-packages/torch/_dynamo/repro/after_dynamo.py", line 117, in debug_wrapper
    compiled_gm = compiler_fn(gm, example_inputs)
  File "/root/minicondaa3/envs/nerfstudio3/lib/python3.8/site-packages/torch/__init__.py", line 1607, in __call__
    return self.compiler_fn(model_, inputs_, **self.kwargs)
torch._dynamo.exc.BackendCompilerFailed: backend='eager' raised:
TypeError: eager() got an unexpected keyword argument 'mode'

Set TORCH_LOGS="+dynamo" and TORCHDYNAMO_VERBOSE=1 for more information

You can suppress this exception and fall back to eager by setting:
    import torch._dynamo
    torch._dynamo.config.suppress_errors = True

(nerfstudio3) root@eiyike-Precision-5820-Tower-X-Series:/home/eiyike/NERF_BRIDGE_WRKSPACE/nerf_bridge

javieryu, 2024-02-27:
can you verify your nerfstudio is installed correctly? This is a torch compiler error so the issue is unrelated to nerfbridge.

eiyike123, 2024-02-27:
I am running the setup from a docker container with ubuntu 20.04 and i am using ros noetic. Its training now but i can view it realtime

eiyike123, 2024-02-27:
> can you verify your nerfstudio is installed correctly? This is a torch compiler error so the issue is unrelated to nerfbridge.

It was installed correctly. and its training now but i cant view it realtime
[image]
[image]

javieryu, 2024-02-27:
That seems like a docker issue then. Most likely the network is not being shared correctly with the host.

eiyike123, 2024-03-09:
What is the best way to
Solve it