javieryu/nerf_bridge #33: Hello!Is the ORB-SLAM3 algorithm currently available on the ROS 2-based NerfBridge?

Allen070206, 2024-05-01:
[image]
From the code, I noticed your comment. Currently, if I use the ORB-SLAM3 algorithm, do I need to modify the code?

javieryu, 2024-05-01:
I haven't actually tried using any of the ORBSLAM3 ROS2 packages -- although I know a couple exist. If you get them working then you shouldn't need to modify any of the code, just pass in a new ``slam_method`` string. The ``slam_method`` string is used to indicate which coordinate convention is used to define the rotation of the camera in the slam frame. You'll just have to double check the coordinate frame of the Odometry messages, and choose the appropriate ``slam_method`` that way. Eventually I'd like to make that whole convention system more transparent, but until then you're welcome to just post in this thread and I will try to help you where I can!

javieryu, 2024-05-20:
Closing this for now, but feel free to reopen if you have more trouble.

closed 2024-05-20 by javieryu