javieryu/nerf_bridge #36: ns-export

eiyike123, 2024-05-30:
Is it possible to export the mesh or point cloud after reconstruction.

javieryu, 2024-05-31:
At the moment I haven't had the chance to modify the scripts for exporting from nerfbridge. It should be easy enough to do, and if you end up implementing it then I would greatly appreciate a PR.

closed 2024-05-31 by javieryu

eiyike123, 2024-05-31:
I tried implementing it but the shape is the problem. The model is 512 ,32 but when I export it it’s 0,32

javieryu, 2024-06-03:
Yeah, that is because the default saving method expects the data to be saved on disk, but nerfbridge doesn't ever write to disk so that data is never there. Therefore when the export script calls `next_train` the dataloader yields nothing. That is the functionality that you will have to implement.

joejep, 2024-06-15:
RuntimeError: Error(s) in loading state_dict for NerfactoModel:
        size mismatch for field.embedding_appearance.embedding.weight: copying a param with shape torch.Size([500, 32]) from checkpoint, the shape in current model is torch.Size([0, 32]).
        

please can you please explain to me what you I should save. A code snippet would be highly appreciated.