javieryu/nerf_bridge #37: GSPLAT Torch Module Missing

CharlieRusso001, 2024-06-10:
Hello, when trying to train the the NeRF Bridge model using the command given to train the ROS2 bag I get an error that looks like the one pasted below.  To install GSPLAT I used the recommended pip command to install the repo from source is there any way to resolve this?

(nerfbridge) charl@charl-XPS-8950:~$ ns-train ros-depth-nerfacto --data configs/desk.json --pipeline.datamanager.use-compressed-rgb True --pipeline.datamanager.dataparser.scene-scale-factor 0.5 --pipeline.datamanager.data-update-freq 8.0
Traceback (most recent call last):
  File "/home/charl/anaconda3/envs/nerfbridge/bin/ns-train", line 5, in <module>
    from nerfstudio.scripts.train import entrypoint
  File "/home/charl/anaconda3/envs/nerfbridge/lib/python3.10/site-packages/nerfstudio/scripts/train.py", line 62, in <module>
    from nerfstudio.configs.method_configs import AnnotatedBaseConfigUnion
  File "/home/charl/anaconda3/envs/nerfbridge/lib/python3.10/site-packages/nerfstudio/configs/method_configs.py", line 50, in <module>
    from nerfstudio.engine.trainer import TrainerConfig
  File "/home/charl/anaconda3/envs/nerfbridge/lib/python3.10/site-packages/nerfstudio/engine/trainer.py", line 45, in <module>
    from nerfstudio.viewer.viewer import Viewer as ViewerState
  File "/home/charl/anaconda3/envs/nerfbridge/lib/python3.10/site-packages/nerfstudio/viewer/viewer.py", line 35, in <module>
    from nerfstudio.models.splatfacto import SplatfactoModel
  File "/home/charl/anaconda3/envs/nerfbridge/lib/python3.10/site-packages/nerfstudio/models/splatfacto.py", line 28, in <module>
    from gsplat._torch_impl import quat_to_rotmat
ModuleNotFoundError: No module named 'gsplat._torch_impl'

javieryu, 2024-06-10:
Can you check the version of gsplat and nerfstudio and post here?

nypyp, 2024-06-13:
same isuue here, the version of gsplat is :
```bash
Found existing installation: gsplat 1.0.0
```

javieryu, 2024-06-13:
Currently Nerfstudio is broken due changes in GSplat 1.0. Reverting to the older version of GSplat should fix the issue. I tested this command and it worked for me yesterday.

`pip install gsplat==0.1.11`

However, it seems to have been fixed in the latest version of Nerfstudio (from a couple days ago). Check this thread out here for more info:

CharlieRusso001, 2024-06-17:
I can confirm that installing the older of gsplat resolves the issue I was having.  I had some ROS2 sourcing errors afterwards but that was most likely unrelated.

closed 2024-06-19 by javieryu

VelvetOddity, 2024-08-01:
> I can confirm that installing the older of gsplat resolves the issue I was having. I had some ROS2 sourcing errors afterwards but that was most likely unrelated.
What's the version of your nerfstudio?