javieryu/nerf_bridge #38: Rendering Results

Augusthyq, 2024-06-12:
Thank you for your good job! I have two questions: 
     1)  When I run orbslam3 calling nerfstudio, only two image frames are displayed, but it seems to be able to render the approximate scene. Is this normal?     
[image]

     2) Can the intrinsic parameters provided by the dataset be copied directly into the JSON file?
Thank you very much if you can help answer.

Augusthyq, 2024-06-12:
> Thank you for your good job! I have two questions: 1) When I run orbslam3 calling nerfstudio, only two image frames are displayed, but it seems to be able to render the approximate scene. Is this [...]

[image]

javieryu, 2024-06-12:
1) Looks like you are using the ROS1 version of NerfBridge. This uses the old Nerfstudio viewer which was unstable, and did not always show all of the poses. From the looks of your TUM results I think you have more than just two images, and sometimes refreshing the page can show the new images (although this is buggy and does not always work).

2) Yes, you can use the camera calibration from the dataset. Just copy the correct values into the JSON.

Augusthyq, 2024-06-12:
Thank you very much for your reply.

closed 2024-06-12 by Augusthyq